lora-aprs/LoRa_APRS_Tracker #5: Display the voltage of the battery of the TTGO T-Beam on the screen

F4EWI, 2020-11-05:
The idea would be to display the TTGO battery voltage and to have the possibility to send this value in the message APRS -> BEACON_MESSAGE. 
It is better to have a voltage displayed rather than an icon which does not give an exact value -> more "professional" as for radiosonde.

peterus, 2020-11-05:
thanks for the ticket, unfortunately I did not had time today, but i will fix it in the next days!
I was thinking of sending the battery voltage as a telemetry data, so aprs.fi can generate graphs etc. what do you think?

raufaser, 2020-11-05:
Please have a look at my pull request. Battery voltage and (dis)charge current are are displayed.
Telemetry is still to-do.

F4EWI, 2020-11-06:
Good idea for the telemetry ...
Piece of code I found in another project on github :
`
#define BATTERY_PIN 35 // battery level measurement pin, here is the voltage divider connected
void getBatteryVoltage()
{
  // we've set 10-bit ADC resolution 2^10=1024 and voltage divider makes it half of maximum readable value (which is 3.3V)
vBat = analogRead(BATTERY_PIN) * 2.0 * (3.3 / 1024.0);
Serial.print("Battery voltage: ");
Serial.print(vBat);
Serial.println("V");
}
`

peterus, 2020-11-06:
Thanks for the Pull Request, I merged it into mainline.

closed 2020-11-06 by peterus