lora-aprs/LoRa_APRS_Tracker #7: Please have a look at my pull request. Battery voltage and (dis)charge current are are displayed.

F4EWI, 2020-11-06:
Please have a look at my pull request. Battery voltage and (dis)charge current are are displayed.
Telemetry is still to-do.

_Originally posted by @raufaser in https://github.com/lora-aprs/LoRa_APRS_Tracker/issues/5#issuecomment-722700742_

I will test this this weekend

peterus, 2020-11-06:
Please report when you have tested and if the changes are okay for you, thanks!

F4EWI, 2020-11-06:
Yes for the tracker the voltage as well as the current is displayed well Thank you
[image]
But I can't see or I don't know how to see the telemetry information?
Maybe this is not available at the moment?
RSSI, SNR, Voltage and currant would be very interesting information to have in telemetry

peterus, 2020-11-06:
telemetry is still not implemented, i will have to do this in the next days.
Voltage and current is possible. RSSI and SNR will be part of an update in the iGate firmware.

peterus, 2021-03-06:
there is an update in #16 where we will send the battery voltage and current flow via plain text in the message, still no telemetry for charts etc.