lora-aprs/LoRa_APRS_Tracker #11: Could you add freq setting on setting.h

faridfarhan28, 2021-02-08:
hi would you like to add frequency setting don setting.g ? beause many country have different regulation on amateur radio bandplan (my country is 432.200Mhz) and newbie don't know how to set frequency on other code. so i hope you can add custom config on setting.h and also SF,BW,CR seeting in setting.h

thanks

faridfarhan28, 2021-02-11:
hi Peterus, i have found this solution but i dont know how to contribute that's program because i'm newbie to use GitHub :D hope you can find the way. thanks

faridfarhan28, 2021-02-11:
i have fork the program but i dont know how to use that hehehe. thanks

peterus, 2021-02-11:
here is the correct place to create tickets 😉
this ticket is completely appreciated as this will be needed in the next time anyway, i am currently busy with the iGate FW, so contributing is highly appreciated 👍

if you are new to github and git i am suggesting to look some youtube videos. a search on youtube for „how to use github“ is bringing a lot of good videos 🙂 and they are very short too!

faridfarhan28, 2021-02-18:
hi i have some edit confiq about this on "pull request" so i wait your approval. 
thanks

closed 2021-02-18 by faridfarhan28

peterus, 2021-02-19:
thanks gor the pull requst! #13 was merged now 😉