lora-aprs/LoRa_APRS_Tracker #12: t-beam t22 1.1 stuck in "INFO Lora init DONE!"

megabyte2010, 2021-02-11:
Can't make this board work. No errors on compile. debug seems ok. Gps led blinks. Pinouts seems to be the same you have in this sketch. The board start ok but keep saying "info, lora init done!". What i am missing here? Thanks.

peterus, 2021-02-11:
are there any other information coming in the log? because this looks like a gps issue.
please try to reset the gps chip with this project: https://github.com/lora-aprs/TTGO-T-Beam_GPS-reset
if this is also not helping: please report back!

megabyte2010, 2021-02-12:
Well, the reset sketch didn't work, but shorting the gps battery did. Now I need to find out how to make pinpoint show tracks with igate without send to aprs is. Thanks.

faridfarhan28, 2021-02-17:
> Well, the reset sketch didn't work, but shorting the gps battery did. Now I need to find out how to make pinpoint show tracks with igate without send to aprs is. Thanks.

I've experienced something like that, change GPS antenna to "Square" and try at outdoor. Device will transmit when GPS LED is blinking or you get 4 GPS more.

megabyte2010, 2021-02-17:
> I've experienced something like that, change GPS antenna to "Square" and try at outdoor. Device will transmit when GPS LED is blinking or you get 4 GPS more.

Well, the problem wasn't the antenna. It's because the gps module wasn't sending any nmea at all. I used the reset sketch but with no effect. Then, I shorted the gps battery, that seems to reset the gps module to factory settings. Maybe the problem is related with the mestastic sketch that I installed some time ago. It's working fine now.

peterus, 2021-02-19:
as i can read that this issue is solved, i will close the ticket 🙂 thanks for the information what you have done @megabyte2010

closed 2021-02-19 by peterus

MAkcanca, 2023-02-22:
I have this particular issue with T-Beam V1.1 and I confirmed with GPS reset sketch that there is data coming in.