lora-aprs/LoRa_APRS_Tracker #18: Enhancement/suggestion - better precision with !DAO! or compressed beacon

LA1HSA, 2021-03-14:
Hi, 
The tracker is working fine and I currently don't have any major gripes with it. However, I have a suggestion about making the positioning a bit more precise.

By utilizing a compressed beacon or by adding !DAO!-data in the comment section you can add greater detail. A more precise position would be very beneficial to SAR operations where clear text APRS data really falls short in precision. Current GNSS modules using three constellations (GPS+Galileo+GLONASS) gives me a reliable sub 1m precision, whereas APRS in its nature is very unprecise.

Please see http://www.aprs.org/aprs12/datum.txt for specifications.

peterus, 2021-05-14:
I can understand that you want more precision, but we need to save somewhere some time on the transmission as LoRa is very slow.
I will add an option to enable or disable this feature in the future.

dj1an, 2021-06-24:
Tried to implement this in #33 
Seems to work, but may be improved.

closed 2021-06-24 by peterus