lora-aprs/LoRa_APRS_Tracker #19: No more settings.h?

F5UJA, 2021-03-19:
Hello,
it seems that the file settings.h in not present anymore in src. Therefore I cannot enter my call and necessary message for the tracker.
Additionnally, I still get compilation errors (arduino pins definition)
Any help highly appreciated!
Thanks

peterus, 2021-03-19:
I have changed the way how to configure the tracker like I am using it for the iGate firmware: via the json file in the data folder: https://github.com/lora-aprs/LoRa_APRS_Tracker/blob/master/data/tracker.json
you have to upload this file via the PlatformIO command "Upload Filesystem Image" found in Visual Studio: "Alien" on the left -> your board -> Platform -> Upload Filesystem Image.

here a screenshot (it is from an older version of platformio):
[image]

regarding the compilation issues with arduino_pin.h:
there is a bug in the last platformio release for the arduino core. make sure you have the latest tracker firmware as we are using a older version (v3.0.0 defined in platformio.ini).
just download the latest tracker code and it should work.

if you have more questions, please just ask ;)

F5UJA, 2021-03-19:
Thanks for helping!
I still have some issues with PIO and with compilation that I must first solve...

F5UJA, 2021-03-21:
Solved! It was an installation issue with PIO.
After re-installation, works like a charm.
Thanks

closed 2021-03-21 by F5UJA