lora-aprs/LoRa_APRS_Tracker #21: Useless info when usb-powered, and without battery pack

dl9sau, 2021-04-03:
I run my device just with USB. -> No battery voltage, current and no charging.

LoRa_APRS_Tracker adds  " - Bat.: 0.00V - Cur.: 0mA" to the comment field.
This info is just useless and costs aitime.

I suggest the following solution:

--- power_management.h.orig	2020-12-13 22:39:19.000000000 +0100
+++ power_management.h	2021-04-03 14:51:00.000000000 +0200
@@ -25,6 +25,8 @@
 	double getBatteryVoltage();
 	double getBatteryChargeDischargeCurrent();

+	bool isBatteryConnect();
+
 private:
 	AXP20X_Class axp;
 };

--- power_management.cpp.orig	2021-04-02 19:02:59.000000000 +0200
+++ power_management.cpp	2021-04-03 15:04:57.000000000 +0200
@@ -84,3 +84,9 @@
 	}
 	return -1.0 * axp.getBattDischargeCurrent();
 }
+
+// cppcheck-suppress unusedFunction
+bool PowerManagement::isBatteryConnect()
+{
+	return axp.isBatteryConnect();
+}

And in LoRa_APRS_Tracker.cpp:
#ifdef  TTGO_T_Beam_V1_0
                if (powerManagement.isBatteryConnect()) {
                        String batteryVoltage(powerManagement.getBatteryVoltage(), 2);
                        String batteryChargeCurrent(powerManagement.getBatteryChargeDischargeCurrent(), 0);

Testet with just usb: works
Not testet: 1. with battery and no usb power. 2. with battery and with usb power.

peterus, 2021-05-14:
there was a fix for this issue here integrated: https://github.com/lora-aprs/LoRa_APRS_Tracker/commit/727c90eeae65a3d0668974cb6f507c4e9ecf07f3

peterus, 2021-05-14:
I will reopen this issue as I have the feeling that i have seen this issue in the last time.
I need to check this again.