lora-aprs/LoRa_APRS_Tracker #22: Altitude 4294967271 feet is a bit high.

dl9sau, 2021-04-03:
DL9SAU-12>APLT00-1,qAU,DJ2DS-10:=5331.03N/00734.14E[352/001/A=4294967271Thomas D23. TTGO-T-Beam LoRa 70cm; tx-only - Bat.: 0.00V - Cur.: 0mA

2^32-4294967271 = 26 feet below NN.

-> Type error.

This function (with unsigned int)
String padding(unsigned int number, unsigned int width)
was called by
                String alt = padding((int)gps.altitude.feet(), 6);

-> negative altitude gets positive.

Also, the large number exceeds what aprs spec wants: fixed 6 bytes length.

I fixed it this way. No diff, because I also made other changes for the aprs message text generation. I'll send the diff seperately.

```
                String alt = String("");
                int alt_int = (int)gps.altitude.feet();
                if (alt_int >= -99999 && alt_int <= 999999) {
                        alt.concat("/A=");
                        if (alt_int < 0)
                                alt = String("-") + padding(alt_int * -1, 5);
                        else
                                alt = padding(alt_int, 6);
                }
```

peterus, 2021-05-14:
@dl9sau I believe we fixed this issue already?

peterus, 2021-05-14:
fixed with https://github.com/lora-aprs/LoRa_APRS_Tracker/commit/727c90eeae65a3d0668974cb6f507c4e9ecf07f3

closed 2021-05-14 by peterus