lora-aprs/LoRa_APRS_Tracker #26: I am missing a step for installation ?

zanco, 2021-04-29:
Hi, 
I have platform io installed on vscode and pointed to the folder .

The README.md states to change my configuration  in data/is-cfg.json

There is however only a tracker.json in the data folder.

The German instruction points to the igate installation and not the LoRa_APRS_Tracker

The French instruction mentions a settings.h in the src folder but there is no settings.h 

I am a bit stuck now.

[image]

Thanks for the help

peterus, 2021-05-14:
@zanco you are right, the correct file is data/tracker.json. I will change the readme!

peterus, 2021-05-14:
fixed in https://github.com/lora-aprs/LoRa_APRS_Tracker/commit/c16d6e43671a9d50a630c860b0de242884633462

closed 2021-05-14 by peterus