lora-aprs/LoRa_APRS_Tracker #29: Support for other OLED drivers?

RipperFox, 2021-06-18:
I shot myself a bit in the foot and accidentally bought an OLED with a SH1106  chip instead of a SSD1306 based one.
At least the pin out allows to solder it directly to my T-Beam (any idea where I can get a SSD1306 based OLED with VCC,GND,SCA,SCL pin out?).
It works if I use the library from https://github.com/nhatuan84/esp32-sh1106-oled
I'm no embedded dev, but shouldn't a more universal OLED library be used in which the display could be configured?

db1sdj, 2021-09-21:
I wanted to reprogram my RDZSonde board with the lora-aprs-Tracker. Me too had the wrong display (the one from RDZSonde: with the color ILI9225 chip) for the tracker SW.
So I am interested too with the possebility of changing the type of display

xe2ad, 2022-08-03:
Hi RipperFox, 
How did you include the SH1106 libraries at Platformio ? I can´t find a correct way to include them at the platformio.ini, so I downloaded them and put the path of the library where it is saved at my computer but neither of both options work.

RipperFox, 2022-08-03:
@xe2ad Huh, that was a while ago :)
I think I just downloaded the lib from https://github.com/nhatuan84/esp32-sh1106-oled and threw it in the src folder and modified the display.cpp a bit - see attached file
[display.zip](https://github.com/lora-aprs/LoRa_APRS_Tracker/files/9249172/display.zip)
t

xe2ad, 2022-08-04:
@RipperFox , Thanks for your reply. In fact I had the libraries from nhatuan84 and just drag them into .pio/libdeps/ttgo-t-bea-v1 folder, seems it solved the way to include them into the tracker project. I replaced the original display.h file with the one you kindly sent and the issues i saw have gone. Now when I try to build the project I get the following error:
[image]

Do you have any clue about what it can be? 
-David.

xe2ad, 2022-08-04:
@RipperFox Many thanks for the display.h file !!!!
I drag the nhatuan84 libraries into the .pio/libdeps/ttgo-t-beam-v1 and that issue has gone.
At the beginning when I used your display.h, I got another error, related with a display_toggle variable... at the end I commented that variable and the display started to work!
Thanks again for your help!
-David.