lora-aprs/LoRa_APRS_Tracker #45: Switch to alternative beacon message with middle button long press

tekk, 2021-11-21:
Hi,
I would love to have the ability to switch to an alternative message (and icon) when holding the middle button.
For example, I'm on a SOTA activation and when returning back I want another beacon message.
Or to switch between pedestrian and car, etc...
Since we have OneButton already included we could use **attachLongPressStop** to trigger between messages.
Does this make any sense?
Waiting for your replies.
BTW I'm willing to help with the code.

dl6nm, 2021-11-22:
I agree with you @tekk.

As a user I'd like to have an option for choosing another callsign/SSID, symbol/overlay and message. Maybe this could be definable in a `profiles` array or object in the `tracker.json` or one `tracker_{SSID}.json` for each `profile`.

tekk, 2021-12-04:
I have implemented required changes in #47 
Please review and let me know.
73 OM7TEK

peterus, 2021-12-19:
implemented in #47

closed 2021-12-19 by peterus