lora-aprs/LoRa_APRS_Tracker #46: ttgo (20210222) t22_1.1 stuck in "INFO Smart Beacon is On"

AlarichvanDalen, 2021-11-30:
TTGO starts with the message
[Info] LoRa init done!
[Info] Smart Beacon is on.
After that, nothing happens. No beacon is sent.
The GPS LED flashes when the device is operated outside.
What ist to do?

dl6nm, 2021-11-30:
I experienced the same behaviour. 
Setting the `debug` flag in `tracker.json` to `false` solved this for me.

AlarichvanDalen, 2021-11-30:
> I experienced the same behaviour. Setting the `debug` flag in `tracker.json` to `false` solved this for me.
Thanks for the tip. Have tested both possible values. No change to the result.

peterus, 2021-12-06:
@AlarichvanDalen @dl6nm please check the serial monitor. It looks like the gps chip has to be resetet. Upload this project and monitor the serial port until you see the nmea output. then re-upload the tracker firmware.
https://github.com/lora-aprs/TTGO-T-Beam_GPS-reset

AlarichvanDalen, 2021-12-07:
I already thought that the GPS is not working properly. I followed the instructions. It looks like the board is now working fine. Thanks for the tip.

closed 2021-12-14 by peterus