lora-aprs/LoRa_APRS_Tracker #48: Support for Heltec HTCC-AB02S

Slayingripper, 2021-12-23:
I'm really loving the project and trying to get this to work on the HTCC-AB02S since it has both GPS and Lora. 
Trying to use the BG_RF95 library is pointless as its not compatible. Is there any documentation on the LORA packet format that needs to be sent out for this to be read by the IGATE ?

peterus, 2021-12-24:
The Tracker Firmware is made for an ESP32 Chip. It will not be so simple to transfer the firmware to other Chips as a lot of ESP depending code is included.
If you want to start your own firmware project for this board, please let me know and i will add an url to this project.

Please give me some days to add a documentation for the protocol. I will write you here again.

@Slayingripper

Slayingripper, 2021-12-24:
I understand that , that is the case, I've been looking into the RadioLib library, which seems to be my closest bet to get this thing working. To clarify, I'm trying to get an APRS tracker working for it to be installed in a hobby rocket for retrieval. The TTGO board is a bit too big to fit. If you know of any smaller but compatible boards that you can suggest, I would appreciate it.

DL9BQ, 2021-12-26:
Jan, DJ1AN built a mostly running version for the AB02S. My first test drive yesterday went well. Will certainly send it to you for testing.

Slayingripper, 2021-12-26:
> 

I would really appreciate it if you could, I'm currently piloting Lora To APRS in Cyprus. I can cover most of the capital city by myself. Looking forwarding to expanding this project as it's quite a cheap way to get onto  the APRS network.

5B4ANU

peterus, 2021-12-26:
@Slayingripper I was talking with Jan, he will publish the source code in one or two days... he found some bugs he want to fix before publishing 😉

dj1an, 2021-12-27:
@Slayingripper 
I just pushed the fork of @peterus Tracker code for the Cubecell Board.
https://github.com/dj1an/Cubecell_LoRa_APRS_Tracker

Talking to the Hardware is a little bit different as you have to use Heltecs LoRaWan_APP in between. 
That's the only way i got it working right now.

Slayingripper, 2021-12-27:
> 

Thank you I've just installed your version, and it seems to be working I will  be testing it further in the coming days.

tekk, 2021-12-30:
@dj1an Can't wait to test your fork, I already ordered 2 boards.
Good job, thanks!