lora-aprs/LoRa_APRS_Tracker #49: Compilliert mit Fehlern.

do2bbc, 2022-01-07:
Hallo zusammen,

bei mir Compilliert der Tracker nicht siehe log. Das I-Gate compilliert ohne probleme.

Was habe ich falsch gemacht?

Vielen Dank für die Hilfe.

73 de DO2BBC

> Executing task: C:\Users\dbw40\.platformio\penv\Scripts\platformio.exe run --environment ttgo-t-beam-v1 <

Processing ttgo-t-beam-v1 (board: ttgo-t-beam; platform: espressif32 @ 3.0.0; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-beam.html
PLATFORM: Espressif 32 (3.0.0) > TTGO T-Beam
HARDWARE: ESP32 240MHz, 1.25MB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.210126 (1.0.4)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep+, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Adafruit GFX Library> 1.7.5
|   |-- <SPI> 1.0
|-- <Adafruit SSD1306> 2.4.0
|   |-- <Adafruit GFX Library> 1.7.5
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|-- <ArduinoJson> 6.17.0
|-- <AXP202X_Library> 1.1.2
|   |-- <Wire> 1.0.1
|-- <LoRa> 0.7.2
|   |-- <SPI> 1.0
|-- <APRS-Decoder-Lib> 0.0.5
|-- <TinyGPSPlus> 1.0.2
|-- <Time> 1.6.1
|-- <OneButton> 1.5.0
|-- <esp-logger> 0.0.1
|-- <SPI> 1.0
|-- <WiFi> 1.0
|-- <Wire> 1.0.1
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <FS> 1.0
Building in release mode
Compiling .pio\build\ttgo-t-beam-v1\src\BeaconManager.cpp.o
Compiling .pio\build\ttgo-t-beam-v1\src\LoRa_APRS_Tracker.cpp.o
In file included from C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/freertos/freertos/FreeRTOSConfig.h:120:0,
                 from C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/freertos/freertos/FreeRTOS.h:99,
                 from C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/Arduino.h:32,
                 from .pio/libdeps/ttgo-t-beam-v1/APRS-Decoder-Lib/src/APRSMessage.h:4,
                 from .pio/libdeps/ttgo-t-beam-v1/APRS-Decoder-Lib/src/APRS-Decoder.h:4,
                 from src/LoRa_APRS_Tracker.cpp:1:
C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/esp32/rom/ets_sys.h:21:21: fatal error: soc/soc.h: No such file or directory
compilation terminated.
Compiling .pio\build\ttgo-t-beam-v1\src\configuration.cpp.o
Compiling .pio\build\ttgo-t-beam-v1\src\display.cpp.o
*** [.pio\build\ttgo-t-beam-v1\src\LoRa_APRS_Tracker.cpp.o] Error 1
In file included from C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/freertos/freertos/FreeRTOSConfig.h:120:0,
                 from C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/freertos/freertos/FreeRTOS.h:99,
                 from C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/Arduino.h:32,
                 from src/configuration.h:7,
                 from src/BeaconManager.h:4,
                 from src/BeaconManager.cpp:1:
C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/esp32/rom/ets_sys.h:21:21: fatal error: soc/soc.h: No such file or directory
compilation terminated.
*** [.pio\build\ttgo-t-beam-v1\src\BeaconManager.cpp.o] Error 1
In file included from C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/freertos/freertos/FreeRTOSConfig.h:120:0,
                 from C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/freertos/freertos/FreeRTOS.h:99,
                 from C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/Arduino.h:32,
                 from .pio/libdeps/ttgo-t-beam-v1/Adafruit GFX Library/Adafruit_GFX.h:5,
                 from src/display.cpp:2:
C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/esp32/rom/ets_sys.h:21:21: fatal error: soc/soc.h: No such file or directory
compilation terminated.
*** [.pio\build\ttgo-t-beam-v1\src\display.cpp.o] Error 1
In file included from C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/freertos/freertos/FreeRTOSConfig.h:120:0,
                 from C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/freertos/freertos/FreeRTOS.h:99,
                 from C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/Arduino.h:32,
                 from C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/libraries/FS/src/FS.h:25,
                 from C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/libraries/SPIFFS/src/SPIFFS.h:17,
                 from src/configuration.cpp:1:
C:/Users/dbw40/.platformio/packages/framework-arduinoespressif32@3.10004.210126/tools/sdk/include/esp32/rom/ets_sys.h:21:21: fatal error: soc/soc.h: No such file or directory
compilation terminated.
*** [.pio\build\ttgo-t-beam-v1\src\configuration.cpp.o] Error 1
========================================================== [FAILED] Took 10.67 seconds ==========================================================

Environment     Status    Duration
--------------  --------  ------------
ttgo-t-beam-v1  FAILED    00:00:10.674
===================================================== 1 failed, 0 succeeded in 00:00:10.674 ===================================================== 
Der Terminalprozess "C:\Users\dbw40\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'ttgo-t-beam-v1'" wurde mit folgendem Exitcode beendet: 1.

Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste.

peterus, 2022-01-21:
ist das noch immer so? schaut nach einem fehler bei der installation aus.
lösche mal diesen Ordner: C:/Users/dbw40/.platformio/ und probiere es noch einmal.

do2bbc, 2022-01-22:
Okay. Habe das ausprobiert und es hat geklappt.
Danke für die Hilfe.

closed 2022-01-22 by peterus