lora-aprs/LoRa_APRS_Tracker #50: Join LoRa_APRS_iGate functionality into a mobile tracker and an attended igate

syystunturi, 2022-01-14:
It's easy to share the WLAN from a mobile phone to the LoRa_APRS_Tracker. It might be a nice feature if one could enable [LoRa_APRS_iGate](https://github.com/lora-aprs/LoRa_APRS_iGate) functionality as well in the tracking device should the APRS tracker happen to have a functioning Internet connection via the predefined WLAN settings.

tekk, 2022-01-15:
On the iGATE you can enter several WIFI profiles so you might only need to add your hotspot's SSID and password, and the iGATE will reconnect after reset (just guessing). This functionality already works on the iGATE firmware.

peterus, 2022-01-21:
the two projects are already to far away to be merged together again.
this would be a lot of work and is definitely nothing for my spare time.

peterus, 2022-05-10:
the iGate firmware is now supporting a mode in which the gps can be used to set the iGate position. hope this is helping more.

closed 2022-05-10 by peterus