lora-aprs/LoRa_APRS_Tracker #52: Problem with date

iw9grl, 2022-01-15:
Hi

after GPS fix, the time is et correctly ( UTC ) but the date remain to 31/12/1999 also after several hour .

could you please help me ?
thank you
73

peterus, 2022-01-21:
you are not the only one who is having issues like this.
currently there is no fix for this as the data is coming directly from the gps library. Maybe it is the gps chip or the library.

efiten, 2022-04-19:
Is there already a solution, or workaround for this ?
With other programs, GPS can read out and show the date correctly.

73
Erwin

stroobandt, 2022-04-19:
The TTGO T-Beam v1.1 appears to be using the NEO-M8N GPS chip instead of the NEO-6M.
This may be the reason for the faulty date indication.

[image]