lora-aprs/LoRa_APRS_Tracker #53: Extend battery runtime

masterkey2000, 2022-01-31:
It would save battery power if the display would turn off after a certain time (e.g. 20 seconds) until you press the on/off button.

snwvo7, 2022-02-11:
Would be nice to have that functionality, the display is 24hrs on so far. Doesn't make any sense.

peterus, 2022-05-10:
#59 is already a good improvement. of course a timer would be better.

xynium, 2023-03-16:
I have turn the code to a kind of "state machine". That change many things so it is on another branch.
Power consumption had fall to 22mA or less

HrSh3rl0ck, 2023-09-08:
Hello, I dont wanna see the display every second - so ive written a few years ago a little Quality of Life Update for me. Ive never published it and now its published and for everyone accessible - now u can de- and activate the display with the IO38 Button. But i dont think it will safe much energy.

Maybe some people will be happy about the update!
You can find the update in my repository or here if the developer Peterus allows the pull request ^^

Best regards, Oliver