lora-aprs/LoRa_APRS_Tracker #67: TTGO in boot loop

uhrbach, 2022-07-02:
Hello,

after flashing the TTGO go in a boot loop.....

Terminal output:

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
I src\LoRa_APRS_Tracker.cpp   79 : LoRa APRS Tracker by OE5BPA (Peter Buchegger)
I src\LoRa_APRS_Tracker.cpp   80 : Version: 22.19.0
E     src\configuration.cpp   31 : Failed to read file, using default configuration.
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40001277  PS      : 0x00060130  A0      : 0x800da34e  A1      : 0x3ffca1f0  
A2      : 0x00000000  A3      : 0x3f401713  A4      : 0x3ffc1ccc  A5      : 0x3ffc1ce0
A6      : 0x400ea600  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffca030  
A10     : 0x3ffca890  A11     : 0x40087f20  A12     : 0x3f40196a  A13     : 0x0000001f
A14     : 0x00000001  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x40001277:0x3ffca1f0 0x400da34b:0x3ffca200 0x400d1f21:0x3ffca220 0x400d20f2:0x3ffca2c0 0x400dc4c7:0x3ffca330 0x400905f5:0x3ffca350
  #0  0x40001277:0x3ffca1f0 in ?? ??:0
  #1  0x400da34b:0x3ffca200 in String::equals(char const*) const at C:\Users\olaf\.platformio\packages\framework-arduinoespressif32\cores\esp32/WString.cpp:857
  #2  0x400d1f21:0x3ffca220 in String::operator==(char const*) const at C:\Users\olaf\.platformio\packages\framework-arduinoespressif32\cores\esp32/WString.h:191
      (inlined by) load_config() at src/LoRa_APRS_Tracker.cpp:354
  #3  0x400d20f2:0x3ffca2c0 in setup() at src/LoRa_APRS_Tracker.cpp:84 (discriminator 8)
  #4  0x400dc4c7:0x3ffca330 in loopTask(void*) at C:\Users\olaf\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:14
  #5  0x400905f5:0x3ffca350 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
Rebooting...
ets Jul 29 2019 12:21:46

dahuafschmied, 2022-07-03:
have you uploaded the filesystem image?

uhrbach, 2022-07-06:
SURE :)
  
Auto-detected: /dev/cu.wchusbserial54260070271
Uploading .pio/build/ttgo-t-beam-v1/firmware.bin
esptool.py v3.0
Serial port /dev/cu.wchusbserial54260070271
Connecting....
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 30:c6:f7:1f:10:08
Uploading stub...
.
.
.
.
 -----------------------------------------
writing at 0x00088000... (100 %)
Wrote 862048 bytes (507861 compressed) at 0x00010000 in 11.8 seconds (effective 584.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
=========================================== [SUCCESS] Took 19.35 seconds ===========================================
Environment     Status    Duration
--------------  --------  ------------
ttgo-t-beam-v1  SUCCESS   00:00:19.350

dahuafschmied, 2022-07-06:
That is the firmware file not the filesystem. 
`Uploading .pio/build/ttgo-t-beam-v1/firmware.bin`

you need to upload the filesystem image
`Uploading .pio/build/ttgo-t-beam-v1/spiffs.bin`

Try:
PlatformIO -> ttgo-t-beam-v1 -> Platform -> Upload Filesystem Image