lora-aprs/LoRa_APRS_Tracker #69: No GPS data

xe2ad, 2022-08-04:
Hi, 
My Tbeam is not getting the position information from GPS, so no position is transmitted. The Tbeam's version is V1.1 with NEO-6M module. I had turned on the tracker all the night expecting to get position info without success.

wolfpcgn, 2022-08-07:
Maybe this can help you: https://github.com/eriktheV-king/TTGO_T-beam_GPS-reset
If the red LED nearby the GPS-module doesn't flash, look at the GPS-antenna its cable or its connector. If you choose the wrong T-BEAM version (0.7 instead of 1.0) there might be no power-supply for the GPS-module.

xe2ad, 2022-08-08:
Thanks for your reply, seems that I was able to run the GPS reset properly, watching the monitor... but still can't get any GPS info. I put the tbeam board in my window, all night without any success (yesterday afternoon it got the date but no position info and the GPS led has been turned off).

ccmolik, 2022-08-09:
For what it's worth, I saw similar behavior with the V1.1 and was convinced the GPS was faulty. It wasn't.

The antenna does seem to be kind of weak, though, and I often find I need to place the device outside with a view of the sky to have it pick up a signal. Being in a window didn't cut it.

This could be coincidental timing but after I added an 18650 battery, the device seemed to keep track of the GPS signal better in my tests, even in a car (red LED blinking next to the GPS chip, etc).

I ended up making a change to the code to have it print the GPS sentences to the console (via the 'Upload and Monitor' feature in PlatformIO):
```
@@ -118,13 +120,13 @@ void loop() {
   if (Config.debug) {
     while (Serial.available() > 0) {
       char c = Serial.read();
-      // Serial.print(c);
+      Serial.print(c);
       gps.encode(c);
     }
   } else {
     while (ss.available() > 0) {
       char c = ss.read();
-      // Serial.print(c);
+      Serial.print(c);
       gps.encode(c);
     }
   }
```

I'd then get the `$GPGSV` sentences in the console, where I could look at the [fourth field](https://docs.novatel.com/OEM7/Content/Logs/GPGSV.htm} to see the number of satellites it saw, even if it's not actually _using_ them until `$GPGGA` fields get populated. This takes a bit as the GPS chip seems to need time to start accepting the satellites it sees.

xe2ad, 2022-08-18:
@wolfpcgn @ccmolik Hi fellows... problem solved. The issue was the GPS antenna. I changed the ceramic antenna and replaced it with an external one and it started to work. 
Thanks for your help !
Best regards,
-David, XE2AD.

closed 2022-08-18 by xe2ad