lora-aprs/LoRa_APRS_Tracker #95: More info needed

toncho11, 2023-02-28:
I have a few questions:
- Can I use two LoRa APRS trackers to communicate between them?
- Can I use Bluetooth to connect each of them to a laptop (one laptop at home with LoRa APRS tracker, one laptop at work with LoRa APRS tracker)? In the  case of laptop there will be software to perform the communication?

richonguzman, 2023-03-20:
It seems you are looking for "Meshtastic" as a solution, as this firmware uses the T-Beam boards and let you do the messaging part your are looking for

o is it that your are looking to do this over/through APRS also?

peterus, 2023-03-22:
Currently this Firmware has no support to send messages over the APRS network. My iGate Firmware is already supporting message forwarding (no creating of messages) from HF to APRS-IS, HF to HF and from APRS-IS to HF.

Support for the Tracker will be soon added.

richonguzman, 2023-03-22:
> The idea is to have an alternative application. So instead of "tracker" to have a "messenger" or "APRS messenger".
> The idea is to have point to point communication.
> I could use it between my work and my home. Send messages from remote parts of the mountain or in case of outage of the 3G/4G networks.
> Please consider this possible development.

I guess this is already done with the Meshtastic firmware for T-Beam boards:

Point to point Message with kiss tnc

Although If you need to go through APRS , you’ll have to mode the script yourself ir wait for someone to do this

closed 2023-09-09 by peterus