lora-aprs/LoRa_APRS_Tracker #101: Lora APRS Tracker update rate

YO3HCM, 2023-04-18:
Hi, congrats for the nice project, I successfully used it with a raspberry pi as GW.

My problem: I need to increase update rate.. ideally 10s between transmissions
No matter how much I lower the numbers in track.json the period will not get lower than 1min.

What is your recommendation?

Thank you

richonguzman, 2023-07-18:
you need to use fixed time delta of the beacons ---> smartBeacon > active = false

smartBeacon uses the values of smart beacon info and not a fixed lower time

peterus, 2023-09-09:
1min is already a lot for lora (you need more than 10 sec to just send your message). if you would decrease it further there would be no airtime for others.

closed 2023-09-09 by peterus