lora-aprs/LoRa_APRS_Tracker #106: T-Beam 1.2 version with AXP2101

Mateus-PP5FMM, 2023-06-12:
Hello,
The new Lilygo T-Beam boards are coming with a new power management chip. In place of the AXP192 is now coming the AXP2101. 
This is causing the APRS LoRA Tracker to malfunction.

[image]

[image]

[image]

peterus, 2023-06-12:
thanks for the info! can you post a link where you bought it? which lora chip is used?

jranma, 2023-07-21:
Same issue here.

Unfortunately, this code will not work on newer T-Beam boards sold by Lilygo (official Aliexpress store)

Since June 2023, AXP192 are no longer produced (or sold ?). New Lilygo T-beam boards are shipped with the new AXP2101.
The battery voltage stays at "0V" and the tracker turns on but has malfunctions.
As it is no longer possible to obtain modules with AXP192 from Lilygo, this problem will concern more and more new users.

Best 73, Julien HB9HRD

Mateus-PP5FMM, 2023-07-21:
> Same issue here.
>
> Unfortunately, this code will not work on newer T-Beam boards sold by Lilygo (official Aliexpress store)
>
> Since June 2023, AXP192 are no longer produced (or sold ?). New Lilygo [...]

Hi Julien!
Try this project:

https://github.com/richonguzman/LoRa_APRS_Tracker 

https://github.com/richonguzman/LoRa_APRS_iGate

From Chile by Ricardo Guzman (CD2RXU)

Very good job by CD2RXU, suports new chip AXP2101.

73!
Mateus PP5FMM

jclifford73, 2023-09-05:
Same problem here in the UK.  I know of 3 users with these new boards that are incompatible.

Anyone know if the CD2RXU trackers are compatible with the OE5BPA igate?

73 John GW4BVE

> > thanks for the info! can you post a link where you bought it? which lora chip is used?
> 
> The LoRa chip is the same, only the power management chip has been changed to the AXP2101. The purchase was made at Lilygo's own store on Aliexpress. I believe this version of the board (V1.2 2023-05-08) is being sold as of now.
> 
> Information about T-Beam V1.2 schematic and new chip AXP2101:
> 
> https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/blob/master/schematic/LilyGo_TBeam_V1.2.pdf
> 
> https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/tree/master/lib/XPowersLib

Mateus-PP5FMM, 2023-09-05:
> Same problem here in the UK. I know of 3 users with these new boards that are incompatible.
>
> Anyone know if the CD2RXU trackers are compatible with the OE5BPA igate?
>
> 73 John GW4BVE

YES! Fully compatible.

73 
Mateus PP5FMM

HrSh3rl0ck, 2023-09-08:
Hello, I've been playing around with LoRa again because a friend destroyed his old T-Beam. And I came across this problem very quickly. Richonguzman's version might work, but the compiler just written so many problems for me. 
I have now adapted the good old software and it has been working for me without any problems for two days. 
Except that the amp display no longer works. For me that's just a "Nice2Have" and so I just removed them.

Maybe some people will be happy about the update!
You can find the update in my repository or if the developer Peterus allows the pull request ^^

Best regards, Oliver (I'm currently still getting the license)

peterus, 2023-09-09:
finally fixed :) 🚀

closed 2023-09-09 by peterus