lora-aprs/LoRa_APRS_Tracker #110: LILYGO T3 v1.6.1 does not work.

sysadminpower2019, 2023-08-29:
I installed the firmware on the LILYGO T3 v1.6.1 and it uploads without any issues. However nothing happens after that, the LEDs flash but the screen is totally black

**Expected behavior**
Should boot and show something on the display

 - Firmware Version: 

**Board name:**
- LILYGO T3 v1.6.1

sysadminpower2019, 2023-08-30:
https://www.aliexpress.com/item/32872078587.html?spm=a2g0o.order_list.order_list_main.5.303d1802nFPDmv

just realized this is not an officially supported board from the Tracker section.

What would be needed to add support to this board, if possible

peterus, 2023-09-09:
This board does not have a GPS build in. you can use it as an iGate. Take a look on my iGate project.

closed 2023-09-09 by peterus