lora-aprs/LoRa_APRS_Tracker #123: LILYGO 433 T-BEAM V1.1 not working

lostdog2, 2023-09-18:
I am new user for T_BEAM. I bought the LILYGO instead TTGO, and is not working. 
The display don't show nothing, so I can't understand if the problem is in the visualization, or the board is totally locked.
I'm not sure if the problem is in the board or in my use of VSCODE (I have not so much practice with it)
First time I tried to compile, I found an error about 'library.json or similar. Searching in internet, I solved deleting the .PIO folder.
This can be the problem?

Many thanks at who will help me!

peterus, 2023-09-19:
is the upload to the board successfully? please look up for some tutorials on how to flash and upload the config. I think this will fix most of your issues.

lostdog2, 2023-09-20:
Peterus, many thanks for you reply.

Yes, the upload is successfully.
I am new to this board and to the upload way (transfer file system).
Before to open this issue, I tried to upload the firmware as file system, or as normal firmware, and all the combination of the two. 
Also, I tried the different version 0,7 and 1.2, despite I have the 1.1 version.
Nothing, the LCD stay blank.
There is a way to understand if the program is running, with blank LCD, or nothing works?
After this, i tried to upload meshtastic using web interface, just for check if the board is working, and all is fine.

Do you think that the compilation error, that I solved deleting the .PIO folder, can be damaged the source code?

Many tanks

lostdog2, 2023-09-20:
Anyway, I will search and watch some tutorial for upload, as you suggested.

Many thanks again

Nerdenator, 2023-09-21:
@lostdog2 I ran into an issue earlier, and like you, ran a web-based upload of Meshtastic firmware successfully. My **guess** is that you'll need to change the version of the firmware you're uploading from v1 to ttgo-t-beam-AXP2101-v1_2. The v1.1 you refer to is on the white label I see on my board with the US FCC symbol. However, next to the SMA header on the board, I also see TBeam-AXP2101-v1.2. This indicates the proper firmware to compile and upload.

lostdog2, 2023-09-21:
@Nerdenator  You are right!
Also my board have two different version..
The upload is working by using 1.2 (but I tried this option to days ago without success, strange..)
Can you say me what is the difference between 'Upload filesystem image' and 'upload'?
The first time I used the 'filesystem' option, as suggested, and it don't work.
The 'upload' only is working..

Many Thanks!!

peterus, 2023-11-25:
`Upload filesystem image`: will upload the configuration (json file)
`upload`: will upload the firmware

both are needed to get a working board. if you change the configuration (json file) you just need to upload the filesystem.

closed 2023-11-25 by peterus