lora-aprs/LoRa_APRS_Tracker #124: Software crashed with T-Beam-v1.1

RWehlingEBK, 2023-09-25:
Thanks for your work but Software crashed with T-Beam-v1.1 
Message from my board:
[INFO][PMU] init done!
[INFO][Main] LoRa APRS Tracker by OE5BPA (Peter Buchegger)
[INFO][Main] Version: 23.36.01
[ERROR][Configuration] Failed to read file, using default configuration.
[INFO][LoRa] Set SPI pins!
[INFO][LoRa] Set LoRa pins!
[INFO][LoRa] frequency: 433775000
[INFO][LoRa] LoRa init done!
[INFO][Main] Smart Beacon is: On
[INFO][Main] setup done...

25.09.2023 18:05:28
Sats: 0 HDOP: 99.99
Next Bcn: ~06:28:15
Bat: 4.13V, 0mA
Smart Beacon: On

Guru Meditation Error: Core 1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core 1 register dump:
PC : 0x400d54ce PS : 0x00060830 A0 : 0x800e2850 A1 : 0x3ffcbb80
A2 : 0xc0bfcbcc A3 : 0x3ffbdb68 A4 : 0x3f403434 A5 : 0x00000000
A6 : 0x3ffc3ca4 A7 : 0x3ffbdb6c A8 : 0x800d54c8 A9 : 0x3ffcbb60
A10 : 0x3ffc3ca4 A11 : 0x00000000 A12 : 0xfffffc02 A13 : 0x000003fe
A14 : 0x7ff00000 A15 : 0x40590000 SAR : 0x0000001d EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000 LBEG : 0x40092599 LEND : 0x400925a9 LCOUNT : 0xffffffff

RWehlingEBK, 2023-09-27:
I found the error in LoRa_APRS_Tracker.cpp at line  363
 i set curr_speed to 1 if curr_speed is o

        if(curr_speed==0)  //RW
                curr_speed=1;
        txInterval = min(BeaconMan.getCurrentBeaconConfig()->smart_beacon.slow_rate,
                         BeaconMan.getCurrentBeaconConfig()->smart_beacon.fast_speed * BeaconMan.getCurrentBeaconConfig()->smart_beacon.fast_rate / curr_speed) *
                     1000;

RWehlingEBK, 2023-09-27:
I found the error in LoRa_APRS_Tracker.cpp at line 363
i set curr_speed to 1 if curr_speed is 0

    if(curr_speed==0)  //RW
            curr_speed=1;
    txInterval = min(BeaconMan.getCurrentBeaconConfig()->smart_beacon.slow_rate,
                     BeaconMan.getCurrentBeaconConfig()->smart_beacon.fast_speed * BeaconMan.getCurrentBeaconConfig()->smart_beacon.fast_rate / curr_speed) *
                 1000;

DJ1MV, 2023-10-30:
Got the same error message and behavior, because I did not upload the configuration file.
All okay afterwards, so please check.

dilmash, 2023-11-10:
> if(curr_speed==0)

I fixed the zerodevide by adding one line after line 354
```C++
    if (BeaconMan.getCurrentBeaconConfig()->smart_beacon.active) {
      // Change the Tx internal based on the current speed
      int curr_speed = (int)gps.speed.kmph();
      if (curr_speed == 0) curr_speed=1;    // <==== make sure it's not zero
```

peterus, 2023-11-25:
I finally checked the source code:
- this line is just called when you are in smart beaconing mode AND
- the current speed is lower than the slow_speed option in your configuration.

my advice: please check your configuration! this should not be a firmware bug! If you defined to be in smart beaconing mode, please add the corresponding configuration definitions!

RWehlingEBK, 2023-11-25:
Thanks Peter,IT was my mistake that caused the (configuration) error. But avoiding the devide by Zero Error is even better. Thanks for ur work!vy73 Ralf DF6VW Gesendet von Outlook für Android

peterus, 2023-11-25:
fixed in mainline :)

closed 2023-11-25 by peterus