lora-aprs/LoRa_APRS_Tracker #134: Issues when Installing on T-BEAM v1.2

sysadminpower2019, 2023-12-07:
Hello

I tried to install this version but also a different project by richonguzman. Unfortunately, neither of them is working for me.

When I use an older version of this project I simply get a screen that says LORA APRS, when I use the current version the screen does not turn on at all.

I did change the call sign

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Debian
 - PlatformIO Version: PlatformIO Core, version 6.1.11
 - Firmware Version: 23.36.0 and others

**Board name:**
- 
 2ASYE-T-Beam

T-Beam-AXP2101-v1.2
**Additional context**
Compiling directly from CLI

meahtr81, 2024-01-03:
I have the same problem with the new Chipset. 
Seems the control of the chipset differs to the old version. 
It hangs up during TX
AXP2101 - Power Managment Chipset
Platform: TTGO TBEAM-AXP2101 V1.2 Production Date labeled next to SMA Connector 20230508.
Visual Studio Code Version : 1.85.1 mit Platform IO

Can i support in some way? Do you need further information?

sm0uqt, 2024-01-23:
did you solve this problem? I have the same board and have had this problem now for a longtime..trying to solve it since november.....im giving up soon and will try to order some other board...

sm0uqt, 2024-01-30:
I solved it by using a old firmware by OE5BPA.

https://github.com/lora-aprs/LoRa_APRS_Tracker

works perfectly now...

sysadminpower2019, 2024-02-20:
I used the version from ricardo
https://github.com/richonguzman/LoRa_APRS_Tracker