lora-aprs/LoRa_APRS_Tracker #145: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

richtigerprashant, 2024-04-02:
Hi, 
I just installed the latest version from the main branch successfully from with pio (console). Now the start screens shows up, display shows after some seconds LoRa initiated and then switches to the overview or dies after a while. Then it goes into an endless loop of restarting as it seems. 

I found this on the monitor console 

--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
[ERROR][Configuration] Failed to read file, using default configuration.
[INFO][LoRa] Set SPI pins!
[INFO][LoRa] Set LoRa pins!
[INFO][LoRa] frequency: 433775000
[INFO][LoRa] LoRa init done!
[INFO][Main] Smart Beacon is: On
[INFO][Main] setup done...
[INFO][Loop] GPS fix state went to VALID
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e5972  PS      : 0x00050031  A0      : 0x400860d0  A1      : 0x3ffbf57c  
A2      : 0x3f800998  A3      : 0x3ffbecb4  A4      : 0x000000ff  A5      : 0x00000ff0  
A6      : 0x3ffbdcc0  A7      : 0x0001e870  A8      : 0x00000000  A9      : 0x3ffbecb0  
A10     : 0x00000001  A11     : 0x3ffc9de0  A12     : 0x00000014  A13     : 0x00000004  
A14     : 0x3ffbd500  A15     : 0x80000001  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x40092074  LEND    : 0x4009207f  LCOUNT  : 0x000013a6  

Backtrace: 0x400e596f:0x3ffbf57c |<-CORRUPTED

  #0  0x400e596f:0x3ffbf57c in uart_ll_get_intsts_mask at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32/include/hal/uart_ll.h:186
      (inlined by) uart_rx_intr_handler_default at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/driver/uart.c:854

ELF file SHA256: 184211336248d8b2

Rebooting...

Anyone knows how to fix this?
Thanks 
Sebastian

richtigerprashant, 2024-04-03:
figured out its happening when the device got a fix and might try to send it goes into the failure.

closed 2024-04-03 by richtigerprashant

dc0sk, 2024-05-05:
I get similar exceptions here, on a T-Beam 1.2

They occur after a  GPS fix is achieved. To me this looks like a memory corruption of some pointer.

Also, they seem to happen in different locations, reproducible with the debug binary:

```
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[INFO][PMU] init done!
[INFO][Main] LoRa APRS Tracker by OE5BPA (Peter Buchegger)
[INFO][Main] Version: 23.36.01
[ERROR][Configuration] Failed to read file, using default configuration.
[INFO][LoRa] Set SPI pins!
[INFO][LoRa] Set LoRa pins!
[INFO][LoRa] frequency: 433775000
[INFO][LoRa] LoRa init done!
[INFO][Main] Smart Beacon is: On
[INFO][Main] setup done...
[INFO][Loop] GPS fix state went to VALID
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e5972  PS      : 0x00050031  A0      : 0x400860d0  A1      : 0x3ffbf57c  
A2      : 0x3f800998  A3      : 0x3ffbecb4  A4      : 0x000000ff  A5      : 0x00000ff0  
A6      : 0x3ffbdcc0  A7      : 0x0001e870  A8      : 0x00000000  A9      : 0x3ffbecb0  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x00000014  A13     : 0x00000004  
A14     : 0x3ffbd500  A15     : 0x80000001  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x40092074  LEND    : 0x4009207f  LCOUNT  : 0x0000159a  

Backtrace: 0x400e596f:0x3ffbf57c |<-CORRUPTED

  #0  0x400e596f:0x3ffbf57c in set_esp_interface_ip(esp_interface_t, IPAddress, IPAddress, IPAddress, IPAddress) at /home/sk/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:147

ELF file SHA256: df03e3874f848237

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[INFO][PMU] init done!
[INFO][Main] LoRa APRS Tracker by OE5BPA (Peter Buchegger)
[INFO][Main] Version: 23.36.01
[ERROR][Configuration] Failed to read file, using default configuration.
[INFO][LoRa] Set SPI pins!
[INFO][LoRa] Set LoRa pins!
[INFO][LoRa] frequency: 433775000
[INFO][LoRa] LoRa init done!
[INFO][Main] Smart Beacon is: On
[INFO][Main] setup done...
[INFO][Loop] GPS fix state went to VALID
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40091ec7  PS      : 0x00060c30  A0      : 0x800e025c  A1      : 0x3ffcbb30  
A2      : 0x3f800ac5  A3      : 0x00000014  A4      : 0x0001e849  A5      : 0x3f800ac5  
A6      : 0x00000039  A7      : 0x34323030  A8      : 0x00000000  A9      : 0x3ffcbb00  
A10     : 0x00000019  A11     : 0xf4310f07  A12     : 0xf4310f07  A13     : 0x3ffc4274  
A14     : 0x00000019  A15     : 0x3f800aac  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000014  LBEG    : 0x40092074  LEND    : 0x4009207f  LCOUNT  : 0x00000000  

Backtrace: 0x40091ec4:0x3ffcbb30 0x400e0259:0x3ffcbb40 0x400e0309:0x3ffcbb60 0x400d49a2:0x3ffcbb80 0x400e2851:0x3ffcbd40

  #0  0x40091ec4:0x3ffcbb30 in __memcpy_aux at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:97
  #1  0x400e0259:0x3ffcbb40 in XPowersAXP2101::setDC5Voltage(unsigned short) at .pio/libdeps/ttgo-t-beam-AXP2101-v1_2/XPowersLib/src/XPowersAXP2101.tpp:1688
      (inlined by) XPowersAXP2101::setPowerChannelVoltage(unsigned char, unsigned short) at .pio/libdeps/ttgo-t-beam-AXP2101-v1_2/XPowersLib/src/XPowersAXP2101.tpp:2989
  #2  0x400e0309:0x3ffcbb60 in XPowersAXP2101::setALDO4Voltage(unsigned short) at .pio/libdeps/ttgo-t-beam-AXP2101-v1_2/XPowersLib/src/XPowersAXP2101.tpp:1891
      (inlined by) XPowersAXP2101::setPowerChannelVoltage(unsigned char, unsigned short) at .pio/libdeps/ttgo-t-beam-AXP2101-v1_2/XPowersLib/src/XPowersAXP2101.tpp:2997
  #3  0x400d49a2:0x3ffcbb80 in loop() at src/LoRa_APRS_Tracker.cpp:341 (discriminator 6)
  #4  0x400e2851:0x3ffcbd40 in Adafruit_SSD1306::drawFastHLine(short, short, short, unsigned short) at .pio/libdeps/ttgo-t-beam-AXP2101-v1_2/Adafruit SSD1306/Adafruit_SSD1306.cpp:721

ELF file SHA256: df03e3874f848237

Rebooting...

```

richtigerprashant, 2024-05-06:
I spent quite some time on fixing this without success. I switched to the version from https://github.com/richonguzman/LoRa_APRS_Tracker and from there its working.

dc0sk, 2024-05-06:
> I spent quite some time on fixing this without success. I switched to the version from https://github.com/richonguzman/LoRa_APRS_Tracker and from there its working.

Thank you for replying, highly appreciated! I switched as well and everything works!

dc0sk, 2024-05-08:
> For me https://github.com/richonguzman/LoRa_APRS_Tracker does not work either it keeps rebooting...
> [Verfolgungsbild][https://github.com/notifications/beacon/ATWXAZPRTOVJQE7OTFP5TCLZBH2KXA5CNFSM6AAAAABFT5YXBSWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT5F3UDM.gif]

Despite this does not belong here, I'll be happy to help you out.

That is most likely because you did not flash the file system with the json config. It is described in chapter 04 of the documentation or here https://randomnerdtutorials.com/esp32-vs-code-platformio-spiffs/

f4exb, 2024-05-08:
yes you have to flash the filesystem and then it works